stereotech/STE-Slicer #28: Лишний слой при нарезке в FULL

ShvetsAA, 2020-12-22:
При нарезке появляется лишний слой, если деталь отодвинуть от основания на 0.1 мм
[image]
[image]

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

frylock34, 2020-12-22:
Добавь конфигурацию глайсера и поделенную модель (находятся в папке Temp)

ShvetsAA, 2020-12-22:
> Добавь конфигурацию глайсера и поделенную модель (находятся в папке Temp)

Файлы в #32

closed 2021-01-18 by frylock34