stereotech/STE-Slicer #61: Объединение мелких разрозненных цилиндрических поддержек

ITorubarov, 2021-11-15:
Проблема: если нависающая часть модели не сплошная, а какая-нибудь фрагментарно-периодическая (как у решётки нейлонового фильтра), то и цилиндрические поддержки печатаются кусочками (рисунок прилагается). Время печати от этого растягивается, грязи обычно становится больше, и выколупывать эти кусочки - просто наказание.
[image]

Решение: объединить мелкие поддержки в один большой и длинный зигзаг, который можно выдирать за один приём.

Можно ли сделать так, чтобы маленькие пустые области между поддержками (меньше некоторой заданной площади, скажем) заполнялись теми же поддержками и рисовались одним зигзагом?

frylock34, 2021-12-01:
передам в Glicer

frylock34, 2022-04-07:
https://tracker.yandex.ru/SLICER-18

closed 2022-04-07 by frylock34