stereotech/STE-Slicer #62: Minimum Infill Area: объём -> площадь

ITorubarov, 2021-11-15:
Есть хорошая настройка Minimum Infill Area, которая задаёт минимальный объём области, который заполняется Infill-ом. Если объём меньше заданного - вместо заполнения печатаются крышки. Очень удобно использовать, чтобы отдельные маленькие элементы модели печатать цельными слоями, без переходов.

Не очень удобно, что это именно объём, а не площадь. На рисунке видно кусочки заполнения, которые было бы очень неплохо печатать одним сплошным крышечным слоем. Но тут настройка в её текущем виде не работает, потому что эти поганые островки Infill-а соединяются с большим материковым Infill-ом выше. Получается, что это вроде как всё один большой объём, и либо во всех слоях надо крыть крышками, либо везде заполнение.
[image]
[image]

Можно ли изменить эту настройку так, чтобы задавалась площадь вместо объёма?
Отдельный вопрос, сохранять ли эту настройку в её текущем виде. Наверное, не нужно - через настройку площади она всё ту же функцию (сплошное заполнение маленьких элементов) будет выполнять. И на глазок площадь прикинуть гораздо проще, чем объём.

frylock34, 2021-12-01:
1. Minimum Infill Area задается в кв.мм, так что это именно площадь
1. Настройка не работает в цилиндрическом режиме.

frylock34, 2022-04-07:
https://tracker.yandex.ru/SLICER-16

closed 2022-04-07 by frylock34