stereotech/STE-Slicer #70: Ошибка: непредсказуемый результат нарезки в classic-режиме без цилиндрического рафта

ITorubarov, 2021-12-17:
При нарезке в режиме classic одной и той же модели с одними и теми же параметрами могут получаться разные варианты G-кода (вроде бы до трёх-четырёх вариантов). Естественно, годный из них только один (с парковкой, установкой температуры, правильной СК G55 и т. д.). В остальных не хватает начальных строчек, иногда вместе с начальными слоями. Иногда на месте все строчки, кроме G0 A0 (впрочем, этого уже достаточно, чтобы голова вдолбилась в модуль).
Цилиндрический рафт везде отключен. Про печать в classic-е с цил. рафтом напишу отдельно.
На предварительном просмотре всегда всё в порядке (если модель вообще нарезалась хоть как-то), недостачи нахожу только в G-коде.
"Сброс" текущей нарезки пробовал делать по-разному (включал-выключал какие-то настройки, менял количество стенок и возвращал обратно, переключал режимы classic-spiral-full) - закономерности с изменением конкретных настроек не заметил.

Здесь уже явно имеется зависимость от модели. Прикрепил три архива:
1) однородный тонкий диск - нарезается вообще без проблем сколько угодно раз
2) дно патрона с отверстиями - нарезал его десять раз подряд, из них три раза модель нарезалась как надо (но без движения G0 A0, т. е. всё-таки не годится), семь раз чего-то не хватало (видно по размерам файлов): итого три разных варианта
3) патрон целиком - нарезал пять раз, один раз был годным, в остальных чего-то не хватает (в одном варианте не хватает много первых строчек, в другом - только движения G0 A0)

[3 патрон целиком.zip](https://github.com/stereotech/STE-Slicer/files/7735405/3.zip)
[1 однородный диск.zip](https://github.com/stereotech/STE-Slicer/files/7735408/1.zip)

[2 дно патрона с отверстиями.zip](https://github.com/stereotech/STE-Slicer/files/7735407/2.zip)

frylock34, 2021-12-21:
Для справки, чтобы сослаться на другую задачу, используй # 
#71

frylock34, 2021-12-21:
Судя по всему идет потеря данных при общении с бекендом. Буду смотреть

closed 2022-04-07 by frylock34