stereotech/STE-Slicer #76: Дополнение: Продумать автоматическое управление вентилятором обдува камеры

ShvetsAA, 2021-12-27:
Не очень удобно на всех принтерах при серийной печати каждый раз включать обдув камеры вручную (в 95% случаев пользоваться им не будут).
Хорошо, если бы была такая настройка в слайсере. А если бы была зависимость от выбранного пластика в слайсере - было бы вообще замечательно (чтобы случайно не включали обдув на высокотемпературных пластиках, если они терпеть не могут обдув)

frylock34, 2021-12-27:
Такое провернуть можно, нужна информация о величине обдува камеры для разных материалов

ShvetsAA, 2021-12-27:
Для начала хватит ручной настройки 0-100% а-ля "обдув модели". Не брать значения обдува модели, а добавить второй, независимый орган управления в слайсере, но с тем же функционалом

frylock34, 2022-04-07:
https://tracker.yandex.ru/SLICER-10

closed 2022-04-07 by frylock34