stereotech/STE-Slicer #78: Доработка настройки перекрытия для плоской и цилиндрической части (full-режим) (Slicer 0-4-3)

ITorubarov, 2021-12-27:
Базовая задача #60 

Перекрытие работает ожидаемым образом. Если в full-режиме печатается монолитная модель, в текущем виде настройка отработает всё как надо.
Если модель имеет стаканообразную форму, при настройке перекрытия в 3D-часть попадают лишние части модели (кроме части дна - ещё вертикальные стойки, которые целиком должны печататься в spiral-е на боковой поверхности оправки).

[image]

[image]

Надо как-то избавиться от лишних кусков в 3D части.
Можно ли для full-режима сделать опциональную возможность воткнуть вспомогательную плоскость на произвольной высоте Z в модели и печатать сердечник не от "стола" (Z=0), а от этой вспомогательной плоскости?
Может быть, это сразу решит и проблему, которая описана в задаче #72 ? Про исчезающий сердечник при отрыве от Z0.

ITorubarov, 2021-12-27:
[image]

Печать сердечника: всё, что выше доп. плоскости попало в область (Base Diameter + Overlap).
Всё, что ниже плоскости - на этом этапе игнорируется.

Печать цилиндрической части: как обычно.

ITorubarov, 2021-12-27:
[image]
[image]

Слушай, по сути, если уронить модель дном на Z0, всё и так нормально работает

frylock34, 2021-12-27:
Только если центрироваться на дальнем от патрона положении

frylock34, 2021-12-27:
> [image]
> [image]
> 
> Слушай, по сути, если уронить модель дном на Z0, всё и так работает, как я хотел.
> Это можно считать допустимым, если цилиндрическая часть модели будет нарезаться изподстольным образом?

Только если центрироваться на дальнем от патрона положении

ITorubarov, 2021-12-27:
> > Слушай, по сути, если уронить модель дном на Z0, всё и так работает, как я хотел.
> > Это можно считать допустимым, если цилиндрическая часть модели будет нарезаться изподстольным образом?
> 
> Только если центрироваться на дальнем от патрона положении

Ну, это понятно.
В принципе, на конце оправки даже удобнее зануляться, чем в начале, не нужно отступ высчитывать.

ITorubarov, 2021-12-27:
Попробую сегодня или завтра таким макаром нарезать и запустить печать. Если принтер никуда не ушатается - задачу на этом закроем, наверное.

frylock34, 2022-04-07:
https://tracker.yandex.ru/SLICER-12

closed 2022-04-07 by frylock34