stereotech/STE-Slicer #85: Изменение настроек по умолчанию (Скорость, Адгезия)

ITorubarov, 2022-03-09:
Проблема 1: в слайсере есть настройки, которые по умолчанию имеют неадекватные значения. При обучении приходится говорить что-то в духе "а вот эту настройку мы изменим так, потому что со значением по умолчанию не получится вообще ничего независимо от модели и материала". Очевидно, назрела необходимость внести изменения.
Вопрос: можно ли сделать так, чтоб при смене режима (Classic -> Full) подгружались другие настройки по умолчанию? Это особенно нужно для настроек рафта (см. ниже).

В таблице по ссылке указаны новые настройки по умолчанию в блоках "Адгезия к столу" и "Скорость". Настройки адгезии написаны для Full-режима, в Classic-режиме всё должно оставаться по-старому!
https://docs.google.com/spreadsheets/d/1czdZL8j4Rd3HFkrxPjFtrBc7fKQ1i7rTLKIW2zJODhY/edit?usp=sharing

ITorubarov, 2022-03-09:
Проблема 2: некоторые настройки Classic Speed и Cylindrical Speed не пересчитываются автоматически при вводе значений в соотв. графу Speed, приходится забивать вручную.

Какие настройки сейчас не пересчитываются автоматически: 
Speed (скорость движения ПГ): Infill Speed, Top/Bottom Speed, Travel Speed
Cooling (скорость вращения вентиляторов обдува модели): Regular Fan Speed, Maximum Fan Speed (ну, т.е. там вообще ничего не пересчитывается)

Какие настройки пересчитываются коряво:
Wall Speed: при записи значения оно же присваивается Outer Wall Speed, оно же * 2 присваивается Inner Wall Speed. Лучше наоборот, Inner Wall Speed = Wall Speed, Outer Wall Speed = Wall Speed / 2.
Initial Layer Travel Speed: как-то высчитывается и от Travel Speed, и от Initial Layer Speed, получается малость кашеобразно. Нужно установить Initial Layer Travel Speed = Initial Layer Speed.

Как должно всё работать:
1) при вводе общей настройки Print Speed - все остальные пересчитываются соотв. образом
2) при вводе каждой отдельной настройки (например, Infill Speed): Classic Speed := Speed, Cylindrical speed := Speed * 1,2

ITorubarov, 2022-03-09:
К вопросу о настройках скорости: по-хорошему, их тоже нужно прописать индивидуально для каждого материала. Но я сейчас не готов список чисел выложить.
Думаю, в каком-то близко обозримом будущем надо прочесать все профили материалов, проверить ещё раз настройки температуры и ретрактов и добавить к ним ещё скорости по умолчанию.

frylock34, 2022-04-07:
https://tracker.yandex.ru/SLICER-5

closed 2022-04-07 by frylock34